iarsystems/iar-vsc-debug #3: Add a Live Watch side bar view

Spaark84, 2022-06-17:
Add a Live Watch side bar view with customizable update interval like it exists in the IAR Embedded Workbench IDE.

MrDanielGerber, 2022-08-23:
Agreed. Probably one of my most used features in EW

zrh-github, 2022-12-13:
Agreed2. I need it so much。

kzoltaan, 2023-01-17:
Agreed3! I want it now! 👍

micaelbo, 2023-04-13:
It is planned, not for next release but hopefully the one after that. No date set yet.

MrHouM, 2023-06-07:
> Add a Live Watch side bar view with customizable update interval like it exists in the IAR Embedded Workbench IDE.

> Add a Live Watch side bar view with customizable update interval like it exists in the IAR Embedded Workbench IDE.

i need it so much，come on as soon as possible，thanks very much，love you！

HampusAdolfsson, 2024-11-26:
The latest release `1.40.1` includes several new views, including a Live Watch window! Please give us any feedback you have on this feature, either here or in a new ticket.

closed 2024-11-26 by HampusAdolfsson