iarsystems/iar-vsc-debug #5: Why is there no response?

olreng, 2022-06-28:
I followed the guide, but there is no response anymore after flash download as shown below.
After a long time, the following error message appears.
"Error: Trying to connect to service with unsupported transport"

---------------------------------------- debug console ----------------------------------------------------
Using C-SPY version: 8.4.3.6413
Driver loaded: c:\Program Files (x86)\IAR Systems\Embedded Workbench 8.4\arm\bin\armjet.dll
Loading the I-jet driver
Loaded macro file: C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.4\arm\config\debugger\ST\STM32F4xx.dmac
Initializing flash loader manager using C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.4\arm\config\flashloader\ST\FlashSTM32F427xI.board
Loading flash module: C:\Work\Project\NGTX\1_Code\fw_interface\NGTX_Interface_FW\EWARM\Application\Debug\Exe\NGTX_Interface_App.out
Flash download warning: 466556 out of 466556 bytes from data record CODE:[0xD1600000,0xD1671E7B] will not be flashed
There were warnings while generating flash loader input.
See the Debug Log window for details.
Loaded macro file: c:\Program Files (x86)\IAR Systems\Embedded Workbench 8.4\arm\config\flashloader\ST\FlashSTM32F4xxx.mac
Loading the I-jet driver
Probe: Probe SW module ver 1.65
Probe: Option: trace(Auto,size_limit=100%)
Probe: Found I-jet, SN=91878
Probe: Opened connection to I-jet:91878
Probe: USB connection verified (7765 packets/sec)
Probe: I-jet, FW ver 7.2, HW Ver:B
Probe: IJET-ARM20 adapter detected
Probe: Versions: JTAG=1.85 SWO=1.40 A2D=1.73 Stream=1.50 SigCom=2.44
Emulation layer version 4.61
SWD clock detected: 12MHz
Notification to init-after-power-up hookup.
Notification to core-connect hookup.
Connected DAP v1 on SWD. Detected DP ID=0x2ba01477.
DMAC: DBGMCU_CR was modified. DBG_SLEEP, DBG_STOP and DBG_STANDBY bits are set.
DMAC: HCLK and FCLK will not be disabled in SLEEP/STOP/STANDBY modes.
Connecting to TAP#0 DAP AHB-AP-CM port 0 (IDR=0x24770011).
Recognized CPUID=0x410fc241 Cortex-M4 r0p1 arch ARMv7-M
Debug resources: 6 instruction comparators, 4 data watchpoints.
CPU status OK
LowLevelReset(system, delay 200)
CPU status - IN RESET
CPU status OK
Loaded debugee: c:\Program Files (x86)\IAR Systems\Embedded Workbench 8.4\arm\config\flashloader\ST\FlashSTM32F4xxxRAM192K_Dual.out
Target reset
Unloaded macro file: c:\Program Files (x86)\IAR Systems\Embedded Workbench 8.4\arm\config\flashloader\ST\FlashSTM32F4xxx.mac
Downloaded C:\Work\Project\NGTX\1_Code\fw_interface\NGTX_Interface_FW\EWARM\Application\Debug\Exe\NGTX_Interface_App.out to flash memory.
1965568 bytes downloaded into FLASH (53.85 Kbytes/sec)
Flash loading completed successfully.
Loaded debugee: C:\Work\Project\NGTX\1_Code\fw_interface\NGTX_Interface_FW\EWARM\Application\Debug\Exe\NGTX_Interface_App.out
Loaded custom formats file: c:\Program Files (x86)\IAR Systems\Embedded Workbench 8.4\arm/config/custom_formats.dat
Loaded custom formats file: C:\Work\Project\NGTX\1_Code\fw_interface\NGTX_Interface_FW\EWARM\Application/custom_formats.dat
The downloaded program doesn't seem to match the expected memory layout of the target system:
Some C variables are placed outside known memory areas:
  os_localtime::lt @ 0xD0481064, os_tm_t
  Del_Info @ 0xD048118C, DEL_INFO
  Delete_GC_Threshold @ 0xD0481259, unsigned char
  EBlockMappingTable @ 0xD046AA2C, EBLOCK_MAPPING_ENTRY[1][488]
  EBlockMappingTableDirtyBitMap @ 0xD04811AC, unsigned char[1][6]
  and more...
Memory is specified as follows:
  0x00000000 - 0x001FFFFF  ROM/Flash
  0x08000000 - 0x081FFFFF  ROM/Flash
  0x10000000 - 0x1000FFFF  RAM
  0x1FFF0000 - 0x1FFF7A3F  ROM/Flash
  0x1FFFC000 - 0x1FFFC03F  ROM/Flash
  0x20000000 - 0x2002FFFF  RAM
  0x22000000 - 0x223FFFFF  RAM
  0x40000000 - 0x4007FFFF  Uncached/SFR
  0x42000000 - 0x42FFFFFF  Uncached/SFR
  0x50000000 - 0x50060FFF  Uncached/SFR
  0x60000000 - 0x9FFFFFFF  Uncached/SFR
  0xA0000000 - 0xA0000FFF  Uncached/SFR
  0xE0000000 - 0xE00FFFFF  Uncached/SFR
C Variables range from 0x10000000 to 0xD1671E7B
LowLevelReset(software, delay 200)
LowLevelReset(system, delay 200)
CPU status - IN RESET
CPU status OK
Download completed.
LowLevelReset(software, delay 200)
Target reset
INFO: Configuring trace using 'Auto,size_limit=100%' setting ...
Trace: Using detected ETMv3CM at address 0xe0041000
Trace: ETMv3CM is not powered-up (ETMCR=0x411)
Trace: ETMv3CM powered-up OK (ETMCR=0xc10)
Trace: Access to detected ETMv3CM(architecture=3.5) initialized (CONF=0x8c842000, CTRL=0xc10, IDR=0x4114f250)
SWO: Manchester, Pin = TDO, Auto divider = 4
INFO: Cannot measure current when I-jet is not powering the target.

---------------------------------------- launch.json ----------------------------------------------------
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "type": "cspy",
            "request": "launch",
            "name": "Application.Debug",
            "target": "arm",
            "program": "${workspaceFolder}\\NGTX_Interface_FW\\EWARM\\Application\\Debug\\Exe\\NGTX_Interface_App.out",
            "driver": "I-jet",
            "stopOnEntry": true,
            "workbenchPath": "${command:iar-config.toolchain}",
            "projectPath": "${workspaceFolder}\\NGTX_Interface_FW\\EWARM\\Application\\Application.ewp",
            "projectConfiguration": "Debug",
            "driverOptions": [
                "--endian=little",
                "--cpu=Cortex-M4",
                "--fpu=VFPv4_SP",
                "-p",
                "C:\\Program Files (x86)\\IAR Systems\\Embedded Workbench 8.4\\arm\\CONFIG\\debugger\\ST\\STM32F439II.ddf",
                "--semihosting",
                "--device=STM32F439II",
                "--multicore_nr_of_cores=1",
                "--jet_standard_reset=4,0,0",
                "--reset_style=\\\"0,-,0,Disabled__no_reset_\\\"",
                "--reset_style=\\\"1,-,0,Software\\\"",
                "--reset_style=\\\"2,-,0,Hardware\\\"",
                "--reset_style=\\\"3,-,0,Core\\\"",
                "--reset_style=\\\"4,-,1,System\\\"",
                "--reset_style=\\\"9,ConnectUnderReset,0,Connect_during_reset\\\"",
                "--jet_power_from_probe=leave_on",
                "--drv_interface=SWD",
                "--jet_cpu_clock=168000000",
                "--drv_catch_exceptions=0xff0",
                "--board_file=C:\\Program Files (x86)\\IAR Systems\\Embedded Workbench 8.4\\arm\\config\\flashloader\\ST\\FlashSTM32F427xI.board"
            ],
            "download": {
                "flashLoader": "C:\\Program Files (x86)\\IAR Systems\\Embedded Workbench 8.4\\arm\\config\\flashloader\\ST\\FlashSTM32F427xI.board",
                "deviceMacros": [
                    "C:\\Program Files (x86)\\IAR Systems\\Embedded Workbench 8.4\\arm\\config\\debugger\\ST\\STM32F4xx.dmac"
                ]
            }
        }
    ]
}

micaelbo, 2022-07-13:
We are aware about the problem which is related to a limitation in the v8 IDE. If possible, please upgrade to the latest EWARM v9.30!

slavek1250, 2022-07-13:
It seems, that PIPE transport is not supported. Are there any plans to support it or SOCKET will be the only available transport?

olreng, 2022-08-03:
ok, i try new version.
it is work very well.
thanks!

closed 2022-08-03 by olreng

mario-pi, 2022-08-03:
> It seems, that PIPE transport is not supported. Are there any plans to support it or SOCKET will be the only available transport?

The client code on the VSCode side can only support sockets. Currently we have no plans to support pipes, was there any specific reason for their use?