iarsystems/iar-vsc-debug #6: Support EWRL78

miervasi-edalab, 2022-07-06:
I love finally seeing support for VSCode from IAR!
It would be awesome if the extension could support other workbenches like EWRL78. I never wrote a VSCode extension myself so I don't know if I could extend it by myself.

HampusAdolfsson, 2022-12-07:
RL78 is now supported (note that you need EWRL78 v5.10 or later).

closed 2022-12-07 by HampusAdolfsson

miervasi-edalab, 2022-12-09:
Hi @HampusAdolfsson, I've tried to launch the debug session. It seems to correctly generate `launch.json` but it early fails with the following message:
```
Using C-SPY version: 8.5.2.7561
Driver loaded: C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.5\rl78\bin\rl78ocd.dll
Reg 'MDAL' not found although registered to a SFR group.
Reg 'MDAH' not found although registered to a SFR group.
C-SPY Processor Descriptor V4.21.4.2472 for RL78
C-SPY OCD Emulator Driver V4.21.3.2447 for RL78
User error: Can not find/open the EXEC file: 'exrl78ocd.dll'.  The file must be located in the 'rl78/config/renesas/execs' directory.
Failed to load debugee: C:\Workspace IAR\mtc_vortice\Debug\Exe\MTC_VORTICE.out
```
I checked the IAR installation directory and the file exrl78ocd.dll is right there in the `rl78/config/renesas/execs` directory. So I guess that something is missing in the `launch.json` file.

micaelbo, 2022-12-09:
Hi @miervasi-edalab. As mentioned above, debugging support comes with EWRL78 v5.10. This release is planned for Q1 2022.

miervasi-edalab, 2022-12-09:
Whops I forgot about that note! Well then I'll wait for the release next year!
Thank you very much for the support!