iarsystems/iar-vsc-debug #8: C-SPY macros

McHighspeed, 2022-07-12:
Hi,
I'm currently trying do download an extra image via the macro __loadImage and therefore created a mac file with the execUserSetup function. Unfortunately I'm not able to download it to my target. I tried the setupMacros option inside the launch json but it will not download the extra image. If I put the mac file inside the deviceMacros option I can see via debug console that the extra image is downloaded to the target but debugging is then not working anymore. I guess it is also the wrong place as the name implies.

Regards

HampusAdolfsson, 2022-07-12:
Hi,
It seems we've made a typo in the launch.json specification. The debugger expects a field called `"macros"`, not "`setupMacros`". I'll make sure we fix this in the next release.

If you change `setupMacros` to `macros` in your launch.json, does that fix your problem? You can ignore any warnings about "Property macros is not allowed".

McHighspeed, 2022-07-12:
`Using C-SPY version: 9.0.8.8418
Driver loaded: C:\Program Files (x86)\IAR Systems\Embedded Workbench 9.0\arm\bin\armstlink.dll
Loading the ST-LINK Driver driver
Loaded macro file: C:\Program Files (x86)\IAR Systems\Embedded Workbench 9.0\arm\config\debugger\ST\STM32F0xx.dmac
Initializing flash loader manager using C:\Program Files (x86)\IAR Systems\Embedded Workbench 9.0\arm\config\flashloader\ST\FlashSTM32F09xxC.board
Loading flash module: XXX\YYY.elf
Loaded macro file: C:\Program Files (x86)\IAR Systems\Embedded Workbench 9.0\arm\config\flashloader\ST\FlashSTM32F0xx.mac
Connected to ST-LINK/V2 [SWD mode 1800 kHz] [Access port: 0] Firmware V2.J39.S7 (Probe no: 55FF71067880565332541467)
Software reset was performed
Initial reset was performed
Loaded debugee: C:\Program Files (x86)\IAR Systems\Embedded Workbench 9.0\arm\config\flashloader\ST\FlashSTM32F0xxxxRAM32K.out
Target reset
Loaded macro file: XX\MAC.mac
Loaded debugee: XXX\YYY.elf
Loaded custom formats file: C:\Program Files (x86)\IAR Systems\Embedded Workbench 9.0\arm/config/custom_formats.dat
Loaded custom formats file: XXX/custom_formats.dat
Software reset was performed
Download completed.
Software reset was performed
Target reset
DMAC: HCLK and FCLK will not be disabled in STOP/STANDBY modes.
Loaded extra image: XXX\ZZZ.elf, image ID 3
Session started
Using 'auto' breakpoint type.
`

it is now downloading to the target but the debugging does not work anymore. Its not halting at main or at startup entry point. Tried the same setup with iar workbench ide(same out files and same mac file) and it is working like expected

HampusAdolfsson, 2022-07-12:
That sounds like it might be another bug, I'll have a look at it. I'll also see if we could support macro parameters.

I won't be working in the coming weeks, however, so it might be a while until I can work on this.

HampusAdolfsson, 2022-08-17:
Sorry for the delay. Could you clarify what is going wrong when using `__loadImage`? Is the image being downloaded to the wrong address, or is it not being downloaded at all. Is there any relevant output in the debug console?

commit a5b238ca (2022-09-02): VSC-306 Rename 'macros' to 'setupMacros'

Closes #8
closed 2022-09-02 by IAR-build-bot