iarsystems/iar-vsc-debug #13: Feature: Display variables in different formats

sbixl, 2022-09-09:
It would be very nice, if the contents of a variable in the variables view can be displayed in different formats (hex dec, bin,...). The Cortex-Debug [1] plugin has a nice feature to solve this by showing a tooltip while hovering of the accordant variable:

[image]

[1]  https://marketplace.visualstudio.com/items?itemName=marus25.cortex-debug

HampusAdolfsson, 2022-09-13:
This is something we definitely want to support and are looking into. One option would be to add a right click option to each variable that lets you set its display format, or we could add some global setting that changes the display format for all variables.

Due to how our debugger API is shaped, we can't show all formats at the same time like Cortex-Debug without affecting performance too much.

davidanderle, 2023-05-13:
I think GDB-based debuggers use [Output formats](https://ftp.gnu.org/old-gnu/Manuals/gdb/html_node/gdb_54.html) to do this. Essentially, you could add your variable as
`myVar, h`
in the watch window to display it as a hexadecimal value. Is this something that would be easier to implement?

HampusAdolfsson, 2023-05-15:
That would work, but a right-click option on the variables would probably be more intuitive and not much more work to implement.

HampusAdolfsson, 2024-11-26:
The `1.40.1` release adds a new "Format" option when right clicking variables, similar to the Embedded Workbench IDE:
[image]

closed 2024-11-26 by HampusAdolfsson