iarsystems/iar-vsc-debug #15: Attach to running target (I-jet/J-Link)

david-iar, 2022-11-23:
Attach to a running target using the "request" type "attach" in the launch.json seems to not work. Tested by @felipe-iar with both IAR I-jet and Segger J-Link + customer using J-Link.

Please advice if this is supposed to work. If not, please see this is as a feature request.

HampusAdolfsson, 2022-11-28:
The "request" field in the launch.json file is a built-in field that all VS Code debuggers have, even if they do not support it (we currently don't). 

It is fairly straightforward to implement, so it is definitely something we can include in a future release. In the meantime I will add an error message when using the "attach" request.

gcampbell-msft, 2023-02-07:
Commenting to indicate another request for this. Thanks!

HampusAdolfsson, 2023-06-26:
This was just released in 1.30.1. Setting `"leaveTargetRunning": true` in launch.json prevents the debugger from stopping the target at the end of the session, and you can then attach to it with `"request": "attach"`. 

Note that `leaveTargetRunning` is only guaranteed to work with EWARM 9.40 at the moment; YMMV with other versions.

closed 2023-06-26 by HampusAdolfsson