iarsystems/iar-vsc-debug #16: Visual Studio code and IAR extension integration - Error: unable to provide automatic debug configuration

ChohanSensei, 2022-12-28:
Dear IAR Team,
I am in contact with your support team and I have been told that the problem I am facing is a bug and I should report it here.

Problem:
I am trying to integrate IAR extension with Visual code studio and following tutorial provided by IAR (youtube video: https://www.youtube.com/watch?v=LHwjdZte2Hk )

I get following error when I try to Debug (video step at 1:46) i.e. Run  Start Debugging. All step till this point (1:46) are successfully done. 
 
[image]

 
Even after debugging the project directly in Embedded workbench and , restarting both programs and PC,  the error is still persistent. 

I have tried excatly what is stated in github documentation but still no success:
“If the IAR C-SPY Debug extension fails to launch a debug session and asks you to debug the project in the IAR Embedded Workbench IDE first, this means that the required configuration settings files are missing. To generate them, you must open the project in the IAR Embedded Workbench IDE and first start and then close a debug session for each project configuration you want to generate the configuration settings files for.” 
 
Info regarding versions:
1.	IAR Embedded Workbench: version 8.42.1.23878
2.	IAR C-SPY Debug extension: version 1.20.1
3.	Debug probe and target MCU: Probe: I-jet, target MCU: Traveo CYT2B7 (for this project) in general Infineon Traveo and PSoC (Arm based microcontrollers)
Extra Information:
IAR Build settings:
[image]

Installed extensions:
[image]

Project Folder structure:
[image]

Best regards,
Chohan

micaelbo, 2022-12-28:
Hi Chohan. Try with installing the IAR Build extension (https://marketplace.visualstudio.com/items?itemName=iarsystems.iar-build). With this, the IAR C-SPY Debug extension automatically provides debug configurations for your project. (With the IAR Build extension, you can also build your project (.ewp) within VS Code)

ChohanSensei, 2022-12-29:
Hi Micaelbo,

I have already installed IAR Build extension v1.20.1, as can be seen in the second screenshot in my query. 

I have even tried to uninstall, install and restart program as well.

Thanks

micaelbo, 2023-01-02:
It is hard to read on your project folder picture but does the opened folder (in VS Code) have a "settings" folder on the same directory level as the ".vscode" folder? If possible, please upgrade to EWARM v9.30 or newer! Then there is no need to run an initial debug session in EW. VS Code will find out the C-SPY settings anyway.

HampusAdolfsson, 2023-01-03:
The extension logs should help provide some more information. You can find them in the `Output` panel (next to `Debug Console`) by selecting `IAR C-SPY Debug` from the dropdown on the right. There should be a line like this:

```
Failed to generate config from .xcl files: <reason>
```

As @micaelbo says, the 'settings' folder in your project folder contains files used by the VS Code extension to configure the debugger. The files of interest are named like this:
 * `<projectname>`.`<configuration>`.general.xcl
 * `<projectname>`.`<configuration>`.driver.xcl

If you could post the contents of these files here, that would be very helpful for us to reproduce the issue.

ChohanSensei, 2023-01-04:
i. I am bound to use IAR8.42. due to project dependencies
ii. The output panel is empty, no message. The configuration files are created, which I found in subfolder of each core ( I am using multicore controller)
iii. Content of both .xcl files below:
--------------------------------------------------------------------------------
driver.xcl:
--------------------------------------------------------------------------------
"--endian=little" 

"--cpu=Cortex-M0+" 

"--fpu=None" 

"-p" 

"C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.4\arm\CONFIG\debugger\Infineon\CYT2B7_M0+.ddf" 

"--drv_verify_download" 

"--semihosting" 

"--device=CYT2B7_M0+" 

"--drv_mtb_regbase=0xF0003000" 

"--drv_mtb_minramaddr=0xF0010000" 

"--drv_mtb_maxramaddr=0xF0010FFF" 

"--multicore_nr_of_cores=1" 

"--jet_standard_reset=4,0,0" 

"--reset_style=\"0,-,0,Disabled__no_reset_\"" 

"--reset_style=\"1,-,0,Software\"" 

"--reset_style=\"2,-,0,Hardware\"" 

"--reset_style=\"4,-,1,System\"" 

"--reset_style=\"7,Traveo2_CM0p_ResetAndAcquireDevice,0,Reset_and_halt_after_bootloader\"" 

"--drv_communication_log=C:\Users\Chohan\Documents\Chohan_Data\1_OneMC\1_Traveo\3_DemoBoards\SocketBoard-176\SDL\tviibe1m\tools\iar\flash\cm0plus\cspycomm.log" 

"--drv_interface=SWD" 

"--drv_catch_exceptions=0xff0" 

"--drv_trace_settings=ETB" 

"--board_file=C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.4\arm\config\flashloader\Infineon\FlashCYT2_CF1M_WF96K.board" 

--------------------------------------------------------------------------------
general.xcl:
--------------------------------------------------------------------------------
"C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.4\arm\bin\armproc.dll" 

"C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.4\arm\bin\armJET.dll" 

"C:\Users\Chohan\Documents\Chohan_Data\1_OneMC\1_Traveo\3_DemoBoards\SocketBoard-176\SDL\tviibe1m\tools\iar\flash\cm0plus\rev_d\Exe\cm0plus.out" 

--plugin="C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.4\arm\bin\armbat.dll" 

--device_macro="C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.4\arm\config\debugger\Infineon\CYT2B7_M0.dmac" 

--device_macro="C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.4\arm\config\debugger\Infineon\CYTVII_M0.dmac" 

--device_macro="C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.4\arm\config\debugger\Infineon\CYTVII_Common.dmac" 

--flash_loader="C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.4\arm\config\flashloader\Infineon\FlashCYT2_CF1M_WF96K.board" 

----------------------------------------------------------------------------------------

These files are stored in subdirectories instead of parent folder, where ".vscode" folder is. 
path: C:\...\tviibe1m\tools\iar\flash\cm0plus\settings
tviibe1m: project folder where .vscode folder is
.vscode folder contains: iar-vsc.json file

I have tried following and still no success:
i. copied 'setting' folder in project folder, where '.vscode' folder is located
ii. coped all files in project folder, where '.vscode' folder is located

Thanks for your support.

HampusAdolfsson, 2023-01-05:
There should also be logs under `IAR C-SPY Debug`; the ones you posted are logs for the IAR Build extension (i.e. under `IAR Build`).

In most cases you will have one Embedded Workbench project per core (one master and one or more partner/slave projects). By default, the VS Code extension will use the debug configuration from the project selected in the "Extension Configuration" panel. Your cm0plus appears to be configured to debug the Cortex-M0+ core. If you have another project for the Cortex-M4 core, you can select it from the project list dropdown and then launch a debug session as you normally would.

ChohanSensei, 2023-01-11:
There is no "IAR C-SPY Debug" option available. There is "IAR Config Generator" log available with following last few lines:

> Skipping file of unsupported type: c:\...\common\src\drivers\syslib\iar\cy_syslib_iar.s
> Skipping file of unsupported type: c:\...\common\src\startup\iar\startup_cm0plus.s
> Done!
> Generating intellisense information for c:\...\tviibe1m\src\main_cm0plus.c
> WARN: Compiler gave non-zero exit code: 1

HampusAdolfsson, 2023-01-16:
That is odd. The extension logs a message right when it starts, which makes the logs appear. Without the logs it's difficult to tell what is wrong...

Do you have this problem with other projects too, or only with this one?

TomasBarakNXP, 2023-06-20:
Hi, I have the same issue using the VSCode plugin on ubuntu with following error:
[6/20/2023, 10:55:43 AM][Debug] Failed to generate config from build extension: Error: Could not get C-SPY cmdline
[6/20/2023, 10:55:43 AM][Debug] Failed to generate config from .xcl files: Error:

Any ideas how to solve this?

HampusAdolfsson, 2023-06-20:
@TomasBarakNXP
Usually this means that the `*.settings.xcl` or `*.driver.xcl` files are missing. They should be in the `settings` directory in you project directory.

Are you using the newly released BXARM 9.40? If so, you should be able to debug without these files. Is the project loaded in the IAR Build extension, and can you see the project files in the 'Files' view?

TomasBarakNXP, 2023-06-20:
Hi @HampusAdolfsson,
Thanks for you quick answer.
Yes - I am using the new BXARM 9.40, however I am getting following error (or warning):
[image]
Any clue?

HampusAdolfsson, 2023-06-20:
@TomasBarakNXP I was mistaken in my last reply -- the BX series does not yet support debugging, or displaying project files in the IAR Build extension.

The warning you see is incorrect and will be fixed in the next extension version. It should say that **IAR Embedded Workbench** 9.20.4 is required (which is currently not available on linux).