iarsystems/iar-vsc-debug #17: Failed to debug Renesas RL78 Project

RLusseni, 2023-01-02:
Hello IAR team, 

I want to use your IAR C-Spy Debug plugin to debug an RL78 application. The debugging configuration is auto-generated by the IAR Build plugin, both plugings on the latest versions. (IAR Build v1.20.1 and IAR C-SPY Debug v1.20.1)

Trying to debug the EW project results in the following error message:
[image]

The output file "FirstRL78.out" has been generated/built with the IAR Build plugin, does exist at the printed location and is not write-protected. Also building the output in the EW IDE and debugging with the C-SPY plugin results in the same error message. Debugging the file in the EW IDE is working flawlessly. 

Debugging configuration is as followed:
```
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "type": "cspy",
            "request": "launch",
            "name": "FirstRL78.Debug",
            "target": "rl78",
            "program": "${workspaceFolder}\\Debug\\Exe\\FirstRL78.out",
            "driver": "Renesas E2",
            "stopOnSymbol": true,
            "trace": true,
            "workbenchPath": "${command:iar-config.toolchain}",
            "projectPath": "${workspaceFolder}\\FirstRL78.ewp",
            "projectConfiguration": "Debug",
            "driverOptions": [
                "--core",
                "s3",
                "--near_const_location",
                "rom0",
                "--near_const_start",
                "0xF2000",
                "--near_const_size",
                "47.749",
                "-p",
                "C:\\Program Files (x86)\\IAR Systems\\Embedded Workbench 8.5\\rl78\\config\\debugger\\ior5f10bmg.ddf",
                "--double=32",
                "-d",
                "e2",
                "--cspybat_inifile",
                "${workspaceFolder}\\settings\\FirstRL78.dnx"
            ]
        }
    ]
}
```

VS Code Debug Console output is as followed:
```
C-SPY Server 2.0 (IAR IDE Platform 8.5.2.7561/W32, compiled with MSVC 19.0.24215.1 on Apr 27 2021 00:40:34)
Build tag: jenkins_8_5_x_nightly_24
Service registry running on local socket on port: 57305
Also available in serialized form in CSpyServer2-ServiceRegistry.txt
Configuring CSpyServer as multicore session owner.
LibSupportEclipse is missing from C:\\Program Files (x86)\\IAR Systems\\Embedded Workbench 8.5
To client: {"seq":0,"type":"event","event":"output","body":{"category":"console","output":"Using C-SPY version: 8.5.2.7561\n"}}
Using C-SPY version: 8.5.2.7561
To client: {"seq":0,"type":"event","event":"output","body":{"category":"console","output":"Driver loaded: C:\\Program Files (x86)\\IAR Systems\\Embedded Workbench 8.5\\rl78\\bin\\rl78ocd.dll\n"}}
Driver loaded: C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.5\rl78\bin\rl78ocd.dll
To client: {"seq":0,"type":"event","event":"output","body":{"category":"console","output":"Timeline frontend service initialized.\n"}}
Timeline frontend service initialized.
To client: {"seq":0,"type":"event","event":"output","body":{"category":"console","output":"C-SPY Processor Descriptor V4.21.1.2409 for RL78\n"}}
C-SPY Processor Descriptor V4.21.1.2409 for RL78
To client: {"seq":0,"type":"event","event":"output","body":{"category":"console","output":"C-SPY OCD Emulator Driver V4.21.1.2409 for RL78\n"}}
C-SPY OCD Emulator Driver V4.21.1.2409 for RL78
To client: {"seq":0,"type":"event","event":"output","body":{"category":"console","output":"Network license: ***censored*** (STD)\n"}}
Network license: ***censored*** (STD)
To client: {"seq":0,"type":"event","event":"output","body":{"category":"console","output":"User error: Can not find/open the EXEC file: 'exrl78ocd.dll'.  The file must be located in the 'rl78/config/renesas/execs' directory.\n"}}
User error: Can not find/open the EXEC file: 'exrl78ocd.dll'.  The file must be located in the 'rl78/config/renesas/execs' directory.
To client: {"seq":0,"type":"event","event":"output","body":{"category":"console","output":"Failed to load debugee: C:\\Daten\\Projekte\\FirstRL78\\Debug\\Exe\\FirstRL78.out\n"}}
Failed to load debugee: C:\Daten\Projekte\FirstRL78\Debug\Exe\FirstRL78.out
To client: {"seq":0,"type":"response","request_seq":2,"command":"launch","success":false,"message":"CSpyException: Failed to load module (C:\\Daten\\Projekte\\FirstRL78\\Debug\\Exe\\FirstRL78.out)"}
```

Best Regards, 
Rene

micaelbo, 2023-01-02:
Hi @RLusseni. Debugging support comes with the next release of IAR Embedded Workbench for RL78 which is v5.10. Planned for Q1 2023.

RLusseni, 2023-01-02:
So it will be necessary to update the EW IDE and its licensing to make use of the C-SPY plugin?

closed 2023-03-17 by jlonnberg