iarsystems/iar-vsc-debug #19: Enable to switch from one call stack to another

JackHerRrer, 2023-01-30:
I hope this is the right place to report this issue.
I have a project that uses freeRTOS. When debugging it, I cannot switch from the call stack of a task to another. The only call stack I can explore is the one of the task that was active at moment I paused the program. 
When using IAR, jumping from one call stack to another works great. 
I don't know if this is linked or not but, in the "embedded tools: rtos viewer", when I click on the task name, it only adds a variable in the "watch view"
Is it a bug ? Or is it just an unavailabe feature ?

HampusAdolfsson, 2023-02-02:
This is something we might be able to support in the future, by exposing windows from C-SPY's RTOS plugins in VS Code.

HampusAdolfsson, 2025-06-10:
The extension now supports the same RTOS views as in IAR Embedded Workbench. To use them, you must enable the corresponding plugin. For example, to enable the freeRTOS plugin for arm, add the following to your launch.json configuration:
```json
            "plugins": [
                "${command:iar-config.toolchain}/arm/plugins/rtos/FreeRtos/FreeRtosArmPlugin.dll"
            ],
```

closed 2025-06-10 by HampusAdolfsson