iarsystems/iar-vsc-debug #20: Debug not started

devwoojin, 2023-03-03:
IAR ARM 8.40.1

It doesn't work after starting debugging.
I can't get into the main() function.
In the IAR compiler, it works normally.

Debug Console Message

Using C-SPY version: 8.3.2.5988
Driver loaded: C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.3\arm\bin\armXDS2.dll
Loaded macro file: C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.3\arm\config\debugger\TexasInstruments\CC2538.dmac
Initializing flash loader manager using C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.3\arm\config\flashloader\TexasInstruments\FlashCC2538RAM32K.board
Loading flash module: C:\Texas Instruments\TIMAC 1.5.2.43299\Projects\mac\Sample\Coordinator-ota\Normal\Exe\msa_cc2538.out
Loaded macro file: C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.3\arm\config\flashloader\TexasInstruments\FlashCC2538.mac
Connecting to TI XDS110 ( Probe no: GY005239 ) COM[12/13]
Firmware: XDS110 (02.03.00.18) Embed with CMSIS-DAP
TI XDS ARM, device revision: 0x00000001, big endian: false, cache: false, board revision: 0x00000000, driver revision: 0x0B020200
Initial reset was performed
Digital States: Disabled because of no description file
416 bytes downloaded and verified (5.14 Kbytes/sec)
Loaded debugee: C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.3\arm\config\flashloader\TexasInstruments\FlashCC2538RAM32K.out
Target reset
Unloaded macro file: C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.3\arm\config\flashloader\TexasInstruments\FlashCC2538.mac

launch.json
{
  "version": "0.2.0",
  "configurations": [
    {
      "type": "cspy",
      "request": "launch",
      "name": "msa_cc2538.Normal-FFD",
      "target": "arm",
      "program": "${workspaceFolder}\\Normal\\Exe\\msa_cc2538.out",
      "driver": "TI XDS",
      "stopOnSymbol": true,
      "workbenchPath": "${command:iar-config.toolchain}",
      "projectPath": "${workspaceFolder}\\msa_cc2538.ewp",
      "projectConfiguration": "Normal-FFD",
      "driverOptions": [
        "--endian=little",
        "--cpu=Cortex-M3",
        "--fpu=None",
        "-p",
        "C:\\Program Files (x86)\\IAR Systems\\Embedded Workbench 8.3\\arm\\CONFIG\\debugger\\TexasInstruments\\CC2538SF53.ddf",
        "--drv_verify_download",
        "--semihosting",
        "--device=CC2538SF53",
        "--drv_vector_table_base=0",
        "--drv_catch_exceptions=0x000",
        "--drv_swo_clock_setup=72000000,0,1000000",
        "--xds_board_file=C:\\Program Files (x86)\\IAR Systems\\Embedded Workbench 8.3\\arm\\config\\debugger\\TexasInstruments\\xds\\CC2538_XDS110_JTAG.dat"
      ],
      "download": {
        "flashLoader": "C:\\Program Files (x86)\\IAR Systems\\Embedded Workbench 8.3\\arm\\config\\flashloader\\TexasInstruments\\FlashCC2538RAM32K.board",
        "deviceMacros": [
          "C:\\Program Files (x86)\\IAR Systems\\Embedded Workbench 8.3\\arm\\config\\debugger\\TexasInstruments\\CC2538.dmac"
        ]
      }
    }
  ]
}

Clicking the Stop Debugging button displays the message below.
CSpyException: Failed to suspend debugger

HampusAdolfsson, 2023-03-07:
Hi,

It appears a change in extension version 1.20.1 is causing this bug in some older Embedded Workbench versions. I'll work on a fix straight away. In the meantime, you can either:
* Download version 1.10.5 of the extension from here:
https://marketplace.visualstudio.com/_apis/public/gallery/publishers/iarsystems/vsextensions/iar-debug/1.10.5/vspackage
Instructions for installing the file are available here:
https://code.visualstudio.com/docs/editor/extension-marketplace#_install-from-a-vsix

Or

* Upgrade to EWARM 9.10.2 or later

closed 2023-03-09 by IAR-build-bot

Fayne19, 2023-09-19:
> Hi,
>
> It appears a change in extension version 1.20.1 is causing this bug in some older Embedded Workbench versions. I'll work on a fix straight away. In the meantime, you can either:
>
>     * Download [...]

In fact this work for me. I was a same problem "CSpyException: Failed to suspend debugger" which still are available on last IAR-debug C/SPY version 1.30 for a project under IAR Embedded Workbench IDE - Arm 8.40.1

commit fe5346b9 (2023-03-09): VSC-367 Avoid calling getCoreState for single-core sessions

getCoreState doesn't work well for some older workbenches
Closes #20