iarsystems/iar-vsc-debug #21: Breakpoints does not work on STM32F4DISCOVERY and IAR EW v8.50

denis-shienkov, 2023-03-12:
Hi, I tried to debug a simple [application](https://github.com/qbs/qbs/tree/master/examples/baremetal/stm32f4discovery/blueblink) on the `STM32F4DISCOVERY` board using the `IAR EW` toolchain.

My setup is:

* Board: `STM32F4DISCOVERY`
* Debugger: `ST-LINK v2` (integrated on the board)
* IAR EW for ARM v8.50

I have configured the `launch.json` file as following:
```json
        {
            "type": "cspy",
            "request": "launch",
            "name": "cspy-debug",
            "target": "arm",
            "program": "${command:qbs.getProductExecutablePath}",
            "driver": "ST-LINK",
            "stopOnSymbol": "main",
            "workbenchPath": "c:/Program Files (x86)/IAR Systems/Embedded Workbench 8.4",
            "projectPath": "${workspaceFolder}",
            "projectConfiguration": "Debug",
            "trace": true,
            "driverOptions": [
                "--endian=little",
                "--cpu=Cortex-M4",
                "--fpu=VFPv4_SP",
                "-p",
                "c:/Program Files (x86)/IAR Systems/Embedded Workbench 8.4/arm/CONFIG/debugger/ST/STM32F407VG.ddf",
                "--semihosting",
                "--device=STM32F407VG",
                "--drv_interface=SWD",
                "--drv_interface_speed=1800",
                "--drv_swo_clock_setup=168000000,1,0",
                "--stlink_reset_strategy=0,2"
            ],
            "download": {
                "flashLoader": "c:/Program Files (x86)/IAR Systems/Embedded Workbench 8.4/arm/config/flashloader/ST/FlashSTM32F4xxx.board",
                "deviceMacros": [
                    "c:/Program Files (x86)/IAR Systems/Embedded Workbench 8.4/arm/config/debugger/ST/STM32F4xx.dmac"
                ]
            }
        }
```

When I press the `debug` button, then I see that the cursor goto the `main` function. 

But, when I set the break-point and clivk on `continue` button, then nothing happens, seems the ELF-application get stalled or something like that.

The `C-SPY` log are following:

<blockquote>
C-SPY Server 2.0 (IAR IDE Platform 8.4.8.6680, compiled with MSVC 19.0.24215.1 on May 30 2020 04:08:07)
Build tag: jenkins_8_4_x_nightly_68
Service registry running on local socket on port: 49356
Also available in serialized form in CSpyServer2-ServiceRegistry.txt
Configuring CSpyServer as multicore session owner.
To client: {"seq":0,"type":"event","event":"output","body":{"category":"console","output":"Using C-SPY version: 8.4.8.6680\n"}}
Using C-SPY version: 8.4.8.6680
To client: {"seq":0,"type":"event","event":"output","body":{"category":"console","output":"Driver loaded: c:\\Program Files (x86)\\IAR Systems\\Embedded Workbench 8.4\\arm\\bin\\armstlink2.dll\n"}}
Driver loaded: c:\Program Files (x86)\IAR Systems\Embedded Workbench 8.4\arm\bin\armstlink2.dll
To client: {"seq":0,"type":"event","event":"output","body":{"category":"console","output":"Loaded macro file: c:/Program Files (x86)/IAR Systems/Embedded Workbench 8.4/arm/config/debugger/ST/STM32F4xx.dmac\n"}}
Loaded macro file: c:/Program Files (x86)/IAR Systems/Embedded Workbench 8.4/arm/config/debugger/ST/STM32F4xx.dmac
To client: {"seq":0,"type":"event","event":"output","body":{"category":"console","output":"Initializing flash loader manager using c:/Program Files (x86)/IAR Systems/Embedded Workbench 8.4/arm/config/flashloader/ST/FlashSTM32F4xxx.board\n"}}
Initializing flash loader manager using c:/Program Files (x86)/IAR Systems/Embedded Workbench 8.4/arm/config/flashloader/ST/FlashSTM32F4xxx.board
To client: {"seq":0,"type":"event","event":"output","body":{"category":"console","output":"Loading flash module: e:/git/ng/git/123/qbs-git/qbs-borland/examples/baremetal/stm32f4discovery/blueblink/build/iar-8_50_5-arm_debug/debug/stm32f4discovery-blueblink.c5b4f8ce/stm32f4discovery-blueblink.out\n"}}
Loading flash module: e:/git/ng/git/123/qbs-git/qbs-borland/examples/baremetal/stm32f4discovery/blueblink/build/iar-8_50_5-arm_debug/debug/stm32f4discovery-blueblink.c5b4f8ce/stm32f4discovery-blueblink.out
To client: {"seq":0,"type":"event","event":"output","body":{"category":"console","output":"Loaded macro file: c:\\Program Files (x86)\\IAR Systems\\Embedded Workbench 8.4\\arm\\config\\flashloader\\ST\\FlashSTM32F4xxx.mac\n"}}
Loaded macro file: c:\Program Files (x86)\IAR Systems\Embedded Workbench 8.4\arm\config\flashloader\ST\FlashSTM32F4xxx.mac
To client: {"seq":0,"type":"event","event":"output","body":{"category":"console","output":"ST-Link firmware too old\n"}}
ST-Link firmware too old
[... 65 lines cut ...]
To client: {"seq":0,"type":"response","request_seq":9,"command":"threads","success":true,"body":{"threads":[{"id":0,"name":"0: Cortex-M4"}]}}
To client: {"seq":0,"type":"event","event":"stopped","body":{"reason":"entry","threadId":0,"allThreadsStopped":true}}
From client: threads(undefined)
To client: {"seq":0,"type":"response","request_seq":12,"command":"threads","success":true,"body":{"threads":[{"id":0,"name":"0: Cortex-M4"}]}}
From client: threads(undefined)
To client: {"seq":0,"type":"response","request_seq":13,"command":"threads","success":true,"body":{"threads":[{"id":0,"name":"0: Cortex-M4"}]}}
From client: stackTrace({"threadId":0,"levels":1})
From client: stackTrace({"threadId":0,"startFrame":0,"levels":20})
To client: {"seq":0,"type":"response","request_seq":14,"command":"stackTrace","success":true,"body":{"stackFrames":[{"id":1000,"source":{"name":"main.c","path":"E:\\git\\ng\\git\\123\\qbs-git\\qbs-borland\\examples\\baremetal\\stm32f4discovery\\blueblink\\main.c","sourceReference":0},"line":61,"column":1,"name":"main","instructionPointerReference":"0x0000000008000082"},{"id":1001,"line":0,"column":0,"name":"[_call_main + 0xd]","instructionPointerReference":"0x00000000080000d6"}]}}
To client: {"seq":0,"type":"response","request_seq":15,"command":"stackTrace","success":true,"body":{"stackFrames":[{"id":1002,"source":{"name":"main.c","path":"E:\\git\\ng\\git\\123\\qbs-git\\qbs-borland\\examples\\baremetal\\stm32f4discovery\\blueblink\\main.c","sourceReference":0},"line":61,"column":1,"name":"main","instructionPointerReference":"0x0000000008000082"},{"id":1003,"line":0,"column":0,"name":"[_call_main + 0xd]","instructionPointerReference":"0x00000000080000d6"}]}}
From client: evaluate({"expression":"_tx_thread_created_ptr","frameId":1000})
CSpyException: Error (col 1): Unknown or ambiguous symbol. _tx_thread_created_ptr
To client: {"seq":0,"type":"response","request_seq":16,"command":"evaluate","success":false,"message":"CSpyException: Error (col 1): Unknown or ambiguous symbol. _tx_thread_created_ptr"}
From client: evaluate({"expression":"pxReadyTasksLists","frameId":1000})
CSpyException: Error (col 1): Unknown or ambiguous symbol. pxReadyTasksLists
To client: {"seq":0,"type":"response","request_seq":17,"command":"evaluate","success":false,"message":"CSpyException: Error (col 1): Unknown or ambiguous symbol. pxReadyTasksLists"}
From client: evaluate({"expression":"_kernel_thread_info_offsets","frameId":1000})
CSpyException: Error (col 1): Unknown or ambiguous symbol. _kernel_thread_info_offsets
To client: {"seq":0,"type":"response","request_seq":18,"command":"evaluate","success":false,"message":"CSpyException: Error (col 1): Unknown or ambiguous symbol. _kernel_thread_info_offsets"}
From client: evaluate({"expression":"_track_list_k_mutex","frameId":1000})
CSpyException: Error (col 1): Unknown or ambiguous symbol. _track_list_k_mutex
To client: {"seq":0,"type":"response","request_seq":19,"command":"evaluate","success":false,"message":"CSpyException: Error (col 1): Unknown or ambiguous symbol. _track_list_k_mutex"}
From client: continue({"threadId":0})
To client: {"seq":0,"type":"response","request_seq":20,"command":"continue","success":true}
</blockquote>

jlonnberg, 2023-03-13:
HI @denis-shienkov,
It seems like the backend fails to properly install the breakpoint into the target. Have you tried the same project in the workbench just to see that it works there? 
At first glance, it seems like you're missing some options for the debugger, e.g., --stlink_probe=stlinkv2, which might be triggering the issue. The VS-code plugins can also consume the command line for the debugger from the xcl-files that the workbench generate once you open the project in EWARM to give you something to comapre with.

denis-shienkov, 2023-03-13:
@jlonnberg ,

>Have you tried the same project in the workbench just to see that it works there?
Yes, it does work within IAR EW.

I also tried to take the output ELF file generated from the IAR EW, and pass it to the VSCode, no results.

> At first glance, it seems like you're missing some options for the debugger, e.g., --stlink_probe=stlinkv2

It did not help.

>  the xcl-files that the workbench generate once you open the project in EWARM to give you something to comapre with

I see two `*.xcl` files `blueblink.Debug.driver.xcl` and `blueblink.Debug.general.xcl` : 
[settings.zip](https://github.com/IARSystems/iar-vsc-debug/files/10954484/settings.zip)

where the `general` contains this:
<blockquote>
"C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.4\arm\bin\armproc.dll" 

"C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.4\arm\bin\armstlink2.dll" 

"E:\git\ng\git\123\qbs-git\qbs-borland\examples\baremetal\stm32f4discovery\blueblink\Debug\Exe\blueblink.out" 

--plugin="C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.4\arm\bin\armbat.dll" 

--device_macro="C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.4\arm\config\debugger\ST\STM32F4xx.dmac" 

--flash_loader="C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.4\arm\config\flashloader\ST\FlashSTM32F4xxx.board" 
</blockquote>

and the `driver` contains this:
<blockquote>
"--endian=little" 

"--cpu=Cortex-M4" 

"--fpu=VFPv4_SP" 

"-p" 

"C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.4\arm\CONFIG\debugger\ST\STM32F407VG.ddf" 

"--semihosting" 

"--device=STM32F407VG" 

"--drv_interface=SWD" 

"--stlink_reset_strategy=0,0" 

"--drv_swo_clock_setup=168000000,1,2000000" 

"--drv_catch_exceptions=0x000" 

"--drv_debug_ap=0" 
</blockquote>

so, I have added the similar options to the VSCode:
```json
        {
            "type": "cspy",
            "request": "launch",
            "name": "cspy-debug",
            "target": "arm",
            "program": "E:/git/ng/git/123/qbs-git/qbs-borland/examples/baremetal/stm32f4discovery/blueblink/Debug/Exe/blueblink.out",
            "driver": "ST-LINK",
            "stopOnSymbol": "main",
            "workbenchPath": "c:/Program Files (x86)/IAR Systems/Embedded Workbench 8.4",
            "projectPath": "${workspaceFolder}",
            "projectConfiguration": "Debug",
            "trace": true,
            "driverOptions": [
                "--endian=little",
                "--cpu=Cortex-M4",
                "--fpu=VFPv4_SP",
                "-p",
                "c:/Program Files (x86)/IAR Systems/Embedded Workbench 8.4/arm/CONFIG/debugger/ST/STM32F407VG.ddf",
                "--semihosting",
                "--device=STM32F407VG",
                "--drv_interface=SWD",
                "--drv_swo_clock_setup=168000000,1,2000000",
                "--stlink_reset_strategy=0,0",
                "--drv_catch_exceptions=0x000",
                "--drv_debug_ap=0"
            ],
            "download": {
                "flashLoader": "c:/Program Files (x86)/IAR Systems/Embedded Workbench 8.4/arm/config/flashloader/ST/FlashSTM32F4xxx.board",
                "deviceMacros": [
                    "c:/Program Files (x86)/IAR Systems/Embedded Workbench 8.4/arm/config/debugger/ST/STM32F4xx.dmac"
                ]
            }
        }
```
but it did not help too.

jlonnberg, 2023-03-14:
I'll have a look at it and see if I can reproduce it. 
Please ensure that the correct type of breakpoint is used as well, see https://github.com/IARSystems/iar-vsc-debug/blob/master/docs/README.md#BreakpointTypes for reference

closed 2023-03-14 by jlonnberg

denis-shienkov, 2023-03-14:
@jlonnberg ,
> You need to install debug plugins version 1.10.5 instead and it should work as expected.

So, do I need to downgrade the `IAR Denig` extension from current `1.20.2` to `1.10.5` ? 

 > The version of EWARM that you're using is incompatible with the latest set of vs-code extension.
 
 What is EWARM version is compatible then?

jlonnberg, 2023-03-15:
Yes, downgrading the debug extension will fix the problem. 

I've tested the stlink and it should work with EWARM 9.20 and later.