iarsystems/iar-vsc-debug #23: Watch window variable value edit inconsitency

davidanderle, 2023-05-13:
Hello,
I've been having a play around with the watch window in VSCode and I've spotted 2 issues:

1. Although the "Set value" functionality is shown, the last element of an array in the watch window cannot be modified. When you set the element's new value, it simply doesn't update. As a workaround you can update the value in the local/static variables window without an issue.
2. A scalar variable's value cannot be modified in the watch window, only under the local/static variables window.

Here's a quick demonstration:
https://github.com/IARSystems/iar-vsc-debug/assets/17354704/d3d09663-6a80-4632-aeb9-58f59eb05cf8

As a side-note, is there a way to colour numbers with an apostrophe in their digits the same colour as ones without? For example, in the demonstration video above you can see that 999 is green, while 1'000 is grey. This apostrophe number interpretation of C-SPY is extremely useful so I would use this feature over the colour mismatch at any point :)

HampusAdolfsson, 2023-05-15:
Hi,

I will look into the issues you found. Regarding the colouring of values in the watch (and variables) window, VS Code automatically classifies the values into different types depending *only* on the value, and uses this to determine the color. Numbers with apostrophes are classified as generic 'values', not 'numbers'.

A workaround is to override the colors used by VS Code, by placing this in your settings:
```json
    "workbench.colorCustomizations": {
        "debugTokenExpression.number": "#cccccc99",
    }
```
or
```json
    "workbench.colorCustomizations": {
        "debugTokenExpression.value": "#b5cea8",
    }
```

davidanderle, 2023-05-15:
Hi,
Thanks for looking into it! The colour customisation settings worked but as expected <struct>s and <array>s are now also coloured green:
[image]
This still gives a more coherent look to me so I'll set this for my setup.

HampusAdolfsson, 2023-05-16:
Yes, probably some time in June.

commit bd8d9f91 (2023-06-26): Fix not being able to set value of last elem in arrays

Fixes #23
commit bf56e128 (2023-06-26): Implement setExpression request

This is needed by VS Code to set the value of top-level expressions in
the watch window

Fixes #23
closed 2023-06-26 by IAR-build-bot