iarsystems/iar-vsc-debug #26: STLink v2 Debug Problems

MrAlexandrKolesnikov, 2023-06-12:
Hi, I try to start debugging but nothing comes out, the firmware is not written to memory

launch.json
```
{
    "configurations": [
    {
        "type": "cspy",
        "request": "launch",
        "name": "Debug",
        "target": "arm",
        "program": "${workspaceFolder}\\iar\\${command:iar-config.project-configuration}\\Exe\\us_gas.out",
        "stopOnSymbol": "main",
        "stopOnEntry": true,
        "workbenchPath": "${command:iar-config.toolchain}",
        "projectPath": "${command:iar-config.project-file}",
        "projectConfiguration": "${command:iar-config.project-configuration}ss",
        "trace": true,
        "driver": "ST-LINK",
        "driverOptions": [
            "--endian=little",
            "--cpu=Cortex-M4",
            "--fpu=VFPv4_SP",
            "-p",
            "${command:iar-config.toolchain}/arm/CONFIG/debugger/ST/STM32L412CB.ddf",
            "--device=STM32L412CB",
            "--semihosting",
            "--drv_interface_speed=auto",
            "--drv_interface=SWD",
            "--drv_catch_exceptions=0xff0",
            "--drv_swo_clock_setup=180000000,0,2000000",
            "--reset_style=\"9,ConnectUnderReset,1,Connect during reset",
        ],
        "download": {
            "flashLoader": "${command:iar-config.toolchain}/arm/config/flashloader/ST/FlashSTM32L41xxB.board",
            "deviceMacros": [
                "${command:iar-config.toolchain}/arm/CONFIG/debugger/ST/STM32L4xx.dmac"
            ]
        },
        "preLaunchTask": "iar: Build Project"
    }
    ]
}
```
        
        
Debug console autput:

```
C-SPY Server 2.0 (IAR IDE Platform 8.3.4.6127, compiled with MSVC 19.0.24215.1 on Sep 19 2019 01:52:44)
Build tag: jenkins_8_3_x_nightly_37
Service registry running on local socket on port: 62598
Also available in serialized form in CSpyServer2-ServiceRegistry.txt
Configuring CSpyServer as multicore session owner.
To client: {"seq":0,"type":"event","event":"output","body":{"category":"console","output":"Using C-SPY version: 8.3.4.6127\n"}}
Using C-SPY version: 8.3.4.6127
To client: {"seq":0,"type":"event","event":"output","body":{"category":"console","output":"Driver loaded: C:\\Program Files (x86)\\IAR Systems\\Embedded Workbench 8.3\\arm\\bin\\armstlink2.dll\n"}}
Driver loaded: C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.3\arm\bin\armstlink2.dll
To client: {"seq":0,"type":"event","event":"output","body":{"category":"console","output":"Loaded macro file: C:\\\\Program Files (x86)\\\\IAR Systems\\\\Embedded Workbench 8.3/arm/CONFIG/debugger/ST/STM32L4xx.dmac\n"}}
Loaded macro file: C:\\Program Files (x86)\\IAR Systems\\Embedded Workbench 8.3/arm/CONFIG/debugger/ST/STM32L4xx.dmac
To client: {"seq":0,"type":"event","event":"output","body":{"category":"console","output":"Initializing flash loader manager using C:\\\\Program Files (x86)\\\\IAR Systems\\\\Embedded Workbench 8.3/arm/config/flashloader/ST/FlashSTM32L41xxB.board\n"}}
Initializing flash loader manager using C:\\Program Files (x86)\\IAR Systems\\Embedded Workbench 8.3/arm/config/flashloader/ST/FlashSTM32L41xxB.board
To client: {"seq":0,"type":"event","event":"output","body":{"category":"console","output":"Loading flash module: C:\\Users\\a.kolesnikov\\Work\\gasflow\\iar\\Debug\\Exe\\us_gas.out\n"}}
Loading flash module: C:\Users\a.kolesnikov\Work\gasflow\iar\Debug\Exe\us_gas.out
To client: {"seq":0,"type":"event","event":"output","body":{"category":"console","output":"None of the application was placed in flash memory.\n"}}
None of the application was placed in flash memory.
To client: {"seq":0,"type":"event","event":"output","body":{"category":"console","output":"Downloaded C:\\Users\\a.kolesnikov\\Work\\gasflow\\iar\\Debug\\Exe\\us_gas.out to flash memory.\n"}}
Downloaded C:\Users\a.kolesnikov\Work\gasflow\iar\Debug\Exe\us_gas.out to flash memory.
To client: {"seq":0,"type":"event","event":"output","body":{"category":"console","output":"0 bytes downloaded into FLASH (0.00 Kbytes/sec)\n"}}
0 bytes downloaded into FLASH (0.00 Kbytes/sec)
[... 89 lines cut ...]
From client: variables({"variablesReference":1000})
Error: Not supported in EWARM v8.40
To client: {"seq":0,"type":"response","request_seq":20,"command":"variables","success":false,"message":"Error: Not supported in EWARM v8.40"}
From client: setBreakpoints({"source":{"name":"main.c","path":"c:\\Users\\a.kolesnikov\\Work\\gasflow\\src\\main.c"},"lines":[158,284],"breakpoints":[{"line":158},{"line":284}],"sourceModified":false})
Setting breakpoint: _ 1 "EMUL_CODE" "{c:\Users\a.kolesnikov\Work\gasflow\src\main.c}.158.1" 0 0 1 "" 0 "" 0
To client: {"seq":0,"type":"response","request_seq":21,"command":"setBreakpoints","success":true,"body":{"breakpoints":[{"verified":true,"line":158,"column":2,"message":"Code @ main.c:158.2, type: default (auto) \n    0x000091D0           [Fetch           ]  Thumb","source":{"name":"main.c","path":"c:\\Users\\a.kolesnikov\\Work\\gasflow\\src\\main.c"}},{"verified":true,"line":284,"column":8,"message":"Code @ main.c:284.8, type: default (auto) \n    0x000093F4           [Fetch           ]  Thumb","source":{"name":"main.c","path":"c:\\Users\\a.kolesnikov\\Work\\gasflow\\src\\main.c"}}]}}
From client: continue({"threadId":0})
To client: {"seq":0,"type":"response","request_seq":22,"command":"continue","success":true}
From client: pause({"threadId":0})
CSpyException: Failed to suspend debugger
To client: {"seq":0,"type":"response","request_seq":23,"command":"pause","success":false,"message":"CSpyException: Failed to suspend debugger"}
From client: pause({"threadId":0})
CSpyException: Failed to suspend debugger
To client: {"seq":0,"type":"response","request_seq":24,"command":"pause","success":false,"message":"CSpyException: Failed to suspend debugger"}
From client: terminate({"restart":false})
To client: {"seq":0,"type":"response","request_seq":25,"command":"terminate","success":true}
To client: {"seq":0,"type":"event","event":"output","body":{"category":"console","output":"Shutting down C-SPY...\n"}}
Shutting down C-SPY...
Shutting down debug server...
Exiting.
CSpyServer exited with code 0
To client: {"seq":0,"type":"event","event":"terminated"}
From client: disconnect({"restart":false})
To client: {"seq":0,"type":"response","request_seq":26,"command":"disconnect","success":true}
```

jlonnberg, 2023-06-15:
@MrAlexandrKolesnikov Hello! Can you please try to execute the same configuration in the EWARM workbench and see if it works there?

MrAlexandrKolesnikov, 2023-06-15:
@jlonnberg Hello! In IAR workbench everything build and debug work too

jlonnberg, 2023-06-15:
@MrAlexandrKolesnikov 
I see. Which version of EWARM are you using?

MrAlexandrKolesnikov, 2023-06-15:
@jlonnberg 8.40.2

closed 2023-09-19 by jlonnberg