iarsystems/iar-vsc-debug #29: Can it intelligently determine whether to recompile the source code before debugging?

TwoBrushes, 2023-09-12:
Is it possible to achieve the same functionality as IAR Workbench, where when clicking "download and debug" or "restart debug", if there are any changes in the source code, it will first perform a rebuild operation?

jlonnberg, 2023-09-12:
It can be done by adding 

"preLaunchTask": "iar: Rebuild Project"

to your launch.json.

closed 2023-09-12 by jlonnberg

TwoBrushes, 2023-09-12:
> It can be done by adding
> 
> "preLaunchTask": "iar: Rebuild Project"
> 
> to your launch.json.

If these settings are applied, even if the source code is not modified, it will still be compiled before debugging. Is it possible to set it so that if the source code is not modified, it will not be compiled?

jlonnberg, 2023-09-12:
@TwoBrushes  Swap it for "iar: Build Project" which will perform an incremental build instead, i.e., only build if the project is not up to date