iarsystems/iar-vsc-debug #30: how can i change the vsc-debug variable displayed format？ the default format is decimal

TwoBrushes, 2023-09-12:


HampusAdolfsson, 2023-09-13:
This is not supported currently, but we are aware that it's a highly requested feature.

See #13 (closing this as duplicate)

closed 2023-09-13 by HampusAdolfsson

TwoBrushes, 2023-09-13:
> This is not supported currently, but we are aware that it's a highly requested feature.
> 
> See #13 (closing this as duplicate)

How can I view the content of a specific memory address in vsc-debug like in iar?

HampusAdolfsson, 2023-09-13:
VS Code's memory view doesn't have this feature at the moment. It can only inspect memory at the address of specific variables/expressions.

You can get around this by adding an expression to the 'Watch' window that points to the address you want to inspect, and then pressing the 'View Binary Data' button. The following example will let you inspect the memory at address `0x1234`:
[image]