iarsystems/iar-vsc-debug #31: Static section in Variables view spins forever

ejohnson-morningstar, 2023-09-21:
**Describe the bug:**
When debugging an application, I can do all of the typical debugging things -- breakpoints, breakpoint expressions, view local variables, view CPU registers, add variables to Watch, view call stack -- but I haven't been able to get the static variables to load in the Variables view. The dropdown ">" button when clicked turns into a spinning wheel, and it seemingly spins forever. I don't see any output in the debug console when doing so, nor does anything show up after letting it sit for a while (in this case, an hour).

**To Reproduce:**
Unclear at the moment, as it has been the only experience when debugging my application.

**Expected behavior:**
Static variables are visible.

**Actual behavior:**
The dropdown turns into a busywheel and variables are never visible.

**Environment:**

- OS: Windows
- Embedded Workbench: ARM
- Embedded Workbench version: 9.20.2
- VSC Extension version: 1.30.2

**Additional context:**
I'm more than happy to do things here locally to help debug this (not sure if it's possible to have additional debug info printed to the console, for example)!

ejohnson-morningstar, 2023-09-22:
Thanks for the tip!

When I click on the static dropdown, I see:
```
From client: variables({"variablesReference":1006})
```
but I don't see a response ever. Seems like the server just isn't responding with them, for some reason. Would you happen know if there's some maximum limit to how many static variables are supported or something?

HampusAdolfsson, 2023-09-28:
There is no intentional limit to the number of variables, but it's possible that there's a bottleneck somewhere (e.g. an algorithm that doesn't scale well) that imposes a practical limit. Do you have a rough idea of how many variables there are in the application you are debugging?

Just to make sure, does the problem go away with other (smaller) projects, for example if you create a new project?

ejohnson-morningstar, 2023-11-09:
Sorry for the delay.

I'm unfortunately relatively new to the project, so I can't really say with any certainty how many static variables we have.

It sounds like we'll probably just move forward using GDB, since that works with IAR binaries well enough.

HampusAdolfsson, 2023-11-14:
I'll close this for now then, but let us know if you want us to take another look at it.

closed 2023-11-14 by HampusAdolfsson