iarsystems/iar-vsc-debug #32: argument variables (like $TARGET_DIR$) in calls to __system1/2/3 are not working

christianfriessnegg, 2023-11-23:
**Describe the bug:**

In debug-macros:
When you use `__system1(), __system2(), __system3()` calls, argument variables like `$PROJ_DIR$, $TARGET_DIR$`, ect. are generally expanded and working when debugging in IAR Workbench.
However when debugging in VSCode, most of these arguement variables are *not* yet supported.

The below test script shows which ones are working in IAR EW and VSCode.

Currently in VSCode only the following are working:
`$CONFIG_NAME$, $PROJ_DIR$, $PROJ_FNAME$, $TOOLKIT_DIR$.`

**To Reproduce:**

```
// ==============================================================
// NOTE: put this code in your IAR debugger macro file (*.mac):
// ==============================================================

execUserSetup()
{
  __system3("echo COMPILER_ARGS: $COMPILER_ARGS$", &out, &err);
  __message "--> Output: ", out;
  __system3("echo CONFIG_NAME: $CONFIG_NAME$", &out, &err);
  __message "--> Output: ", out;
  __system3("echo CUR_DIR: $CUR_DIR$", &out, &err);
  __message "--> Output: ", out;
  __system3("echo CUR_LINE: $CUR_LINE$", &out, &err);
  __message "--> Output: ", out;
  __system3("echo DATE: $DATE$", &out, &err);
  __message "--> Output: ", out;
  __system3("echo EW_DIR: $EW_DIR$", &out, &err);
  __message "--> Output: ", out;
  __system3("echo EXE_DIR: $EXE_DIR$", &out, &err);
  __message "--> Output: ", out;
  __system3("echo FILE_BNAME: $FILE_BNAME$", &out, &err);
  __message "--> Output: ", out;
  __system3("echo FILE_BPATH: $FILE_BPATH$", &out, &err);
  __message "--> Output: ", out;
  __system3("echo FILE_DIR: $FILE_DIR$", &out, &err);
  __message "--> Output: ", out;
  __system3("echo FILE_FNAME: $FILE_FNAME$", &out, &err);
  __message "--> Output: ", out;
  __system3("echo FILE_PATH: $FILE_PATH$", &out, &err);
  __message "--> Output: ", out;
  __system3("echo LIST_DIR: $LIST_DIR$", &out, &err);
  __message "--> Output: ", out;
  __system3("echo OBJ_DIR: $OBJ_DIR$", &out, &err);
  __message "--> Output: ", out;
  __system3("echo PROJ_DIR: $PROJ_DIR$", &out, &err);
  __message "--> Output: ", out;
  __system3("echo PROJ_FNAME: $PROJ_FNAME$", &out, &err);
  __message "--> Output: ", out;
  __system3("echo PROJ_PATH: $PROJ_PATH$", &out, &err);
  __message "--> Output: ", out;
  __system3("echo TARGET_DIR: $TARGET_DIR$", &out, &err);
  __message "--> Output: ", out;
  __system3("echo TARGET_BNAME: $TARGET_BNAME$", &out, &err);
  __message "--> Output: ", out;
  __system3("echo TARGET_BPATH: $TARGET_BPATH$", &out, &err);
  __message "--> Output: ", out;
  __system3("echo TARGET_FNAME: $TARGET_FNAME$", &out, &err);
  __message "--> Output: ", out;
  __system3("echo TARGET_PATH: $TARGET_PATH$", &out, &err);
  __message "--> Output: ", out;
  __system3("echo TOOLKIT_DIR: $TOOLKIT_DIR$", &out, &err);
  __message "--> Output: ", out;
  __system3("echo USER_NAME: $USER_NAME$", &out, &err);
  __message "--> Output: ", out;
  __system3("echo WS_DIR: $WS_DIR$", &out, &err);
  __message "--> Output: ", out;
}

```

**Expected behavior:**

All of the argument variables should be expanded, except a few which are known to not work in debug-mode.
The calls made with `__system1/2/3` should work because variables (paths) are expanded properly.

Here is debug-log output when run in IAR (working, except those which are known to *NOT* work in debug-mode):

NOTE: if you see line with just the `$VAR_NAME$` as output in one line, you know it could not be evaluated.
Some are expected to fail in debug-mode, IGNORE THOSE: 
`$COMPILER_ARGS$, $CUR_LINE$, $FILE_BNAME$, $FILE_BPATH$, $FILE_DIR$, $FILE_FNAME$, $FILE_PATH$`.

```
...
	Thu Nov 23, 2023 10:31:08: : $COMPILER_ARGS$	
	Thu Nov 23, 2023 10:31:08: : --> Output: COMPILER_ARGS: $COMPILER_ARGS$
	
	Thu Nov 23, 2023 10:31:08: : --> Output: CONFIG_NAME: RAILCON
	
	Thu Nov 23, 2023 10:31:08: : --> Output: CUR_DIR: D:\DEV\qcc710-fw-2.0\SES
	
	Thu Nov 23, 2023 10:31:08: : $CUR_LINE$	
	Thu Nov 23, 2023 10:31:08: : --> Output: CUR_LINE: $CUR_LINE$
	
	Thu Nov 23, 2023 10:31:08: : --> Output: DATE: 23/11/2023
	
	Thu Nov 23, 2023 10:31:08: : --> Output: EW_DIR: "C:\Program Files\IAR Systems\Embedded Workbench 9.2"
	
	Thu Nov 23, 2023 10:31:08: : --> Output: EXE_DIR: D:\DEV\qcc710-fw-2.0\SES\RAILCON\Exe
	
	Thu Nov 23, 2023 10:31:08: : $FILE_BNAME$	
	Thu Nov 23, 2023 10:31:08: : --> Output: FILE_BNAME: $FILE_BNAME$
	
	Thu Nov 23, 2023 10:31:08: : $FILE_BPATH$	
	Thu Nov 23, 2023 10:31:08: : --> Output: FILE_BPATH: $FILE_BPATH$
	
	Thu Nov 23, 2023 10:31:08: : $FILE_DIR$	
	Thu Nov 23, 2023 10:31:08: : --> Output: FILE_DIR: $FILE_DIR$
	
	Thu Nov 23, 2023 10:31:08: : $FILE_FNAME$	
	Thu Nov 23, 2023 10:31:08: : --> Output: FILE_FNAME: $FILE_FNAME$
	
	Thu Nov 23, 2023 10:31:08: : $FILE_PATH$	
	Thu Nov 23, 2023 10:31:08: : --> Output: FILE_PATH: $FILE_PATH$
	
	Thu Nov 23, 2023 10:31:08: : --> Output: LIST_DIR: D:\DEV\qcc710-fw-2.0\SES\RAILCON\List
	
	Thu Nov 23, 2023 10:31:08: : --> Output: OBJ_DIR: D:\DEV\qcc710-fw-2.0\SES\RAILCON\Obj
	
	Thu Nov 23, 2023 10:31:08: : --> Output: PROJ_DIR: D:\DEV\qcc710-fw-2.0\SES
	
	Thu Nov 23, 2023 10:31:08: : --> Output: PROJ_FNAME: QCC_710_SES
	
	Thu Nov 23, 2023 10:31:08: : --> Output: PROJ_PATH: D:\DEV\qcc710-fw-2.0\SES\QCC_710_SES.ewp
	
	Thu Nov 23, 2023 10:31:08: : --> Output: TARGET_DIR: D:\DEV\qcc710-fw-2.0\SES\RAILCON\Exe
	
	Thu Nov 23, 2023 10:31:08: : --> Output: TARGET_BNAME: QCC_710_SES
	
	Thu Nov 23, 2023 10:31:08: : --> Output: TARGET_BPATH: D:\DEV\qcc710-fw-2.0\SES\RAILCON\Exe\QCC_710_SES
	
	Thu Nov 23, 2023 10:31:08: : --> Output: TARGET_FNAME: QCC_710_SES.out
	
	Thu Nov 23, 2023 10:31:08: : --> Output: TARGET_PATH: D:\DEV\qcc710-fw-2.0\SES\RAILCON\Exe\QCC_710_SES.out
	
	Thu Nov 23, 2023 10:31:08: : --> Output: TOOLKIT_DIR: "C:\Program Files\IAR Systems\Embedded Workbench 9.2\arm"
	
	Thu Nov 23, 2023 10:31:08: : --> Output: USER_NAME: frc
	
	Thu Nov 23, 2023 10:31:08: : --> Output: WS_DIR: D:\DEV\qcc710-fw-2.0\SES
...
```

**Actual behavior:**

Here is debug-log output when run in VS-Code (buggy - does not expand most of the variables):

NOTE: if you see line with just the `$VAR_NAME$` as output in one line, you know it could not be evaluated.
You will notice that almost all variables could not be evaluated/expanded.
Only the following are working in VSCode: `$CONFIG_NAME$, $PROJ_DIR$, $PROJ_FNAME$, $TOOLKIT_DIR$`.

```
...
$COMPILER_ARGS$
--> Output: COMPILER_ARGS: $COMPILER_ARGS$
--> Output: CONFIG_NAME: RAILCON
$CUR_DIR$
--> Output: CUR_DIR: $CUR_DIR$
$CUR_LINE$
--> Output: CUR_LINE: $CUR_LINE$
$DATE$
--> Output: DATE: $DATE$
$EW_DIR$
--> Output: EW_DIR: $EW_DIR$
$EXE_DIR$
--> Output: EXE_DIR: $EXE_DIR$
$FILE_BNAME$
--> Output: FILE_BNAME: $FILE_BNAME$
$FILE_BPATH$
--> Output: FILE_BPATH: $FILE_BPATH$
$FILE_DIR$
--> Output: FILE_DIR: $FILE_DIR$
$FILE_FNAME$
--> Output: FILE_FNAME: $FILE_FNAME$
$FILE_PATH$
--> Output: FILE_PATH: $FILE_PATH$
$LIST_DIR$
--> Output: LIST_DIR: $LIST_DIR$
$OBJ_DIR$
--> Output: OBJ_DIR: $OBJ_DIR$
--> Output: PROJ_DIR: D:\DEV\qcc710-fw-2.0\SES
--> Output: PROJ_FNAME: QCC_710_SES.ewp
$PROJ_PATH$
--> Output: PROJ_PATH: $PROJ_PATH$
$TARGET_DIR$
--> Output: TARGET_DIR: $TARGET_DIR$
$TARGET_BNAME$
--> Output: TARGET_BNAME: $TARGET_BNAME$
$TARGET_BPATH$
--> Output: TARGET_BPATH: $TARGET_BPATH$
$TARGET_FNAME$
--> Output: TARGET_FNAME: $TARGET_FNAME$
$TARGET_PATH$
--> Output: TARGET_PATH: $TARGET_PATH$
--> Output: TOOLKIT_DIR: "C:\Program Files\IAR Systems\Embedded Workbench 9.2\arm"
$USER_NAME$
--> Output: USER_NAME: $USER_NAME$
$WS_DIR$
--> Output: WS_DIR: $WS_DIR$
...
```

**Environment:**

- OS: [Windows 10]
- Embedded Workbench: [Arm]
- Embedded Workbench version: 9.40.1
- VSC Extension version: IAR CSPY-Debug v1.30.3

micaelbo, 2023-11-27:
Now listed to be investigated for the next extension release.

HampusAdolfsson, 2024-11-26:
This has been fixed, but it required changes in the C-SPY backend code and thus cannot be released as an extension update. The fix will be released with the next EWARM release instead.

closed 2024-11-26 by HampusAdolfsson