iarsystems/iar-vsc-debug #34: Resuming from breakpoints broken with FreeRTOS

cosberg-rzl, 2024-01-30:
**Describe the bug:**
Setting breakpoints seems to (mostly) work, however resuming from them doesn't appear to work at all. The editor freezes for a bit, then re-enters the same breakpoint you were just at. This happens for "step over", "step into", and "continue". The app is essentially stuck in this state until you reset it externally. While stuck here, most other features seem to generally work. There's an error in the debug console that appears whenever you try to continue from this state:
`Error: Unknown or ambiguous symbol. _tx_thread_created_ptr`

**To Reproduce:**
Probably difficult without our project (which we of course can't provide). The steps are pretty simple though, just setting a breakpoint is enough.

**Environment:**

- OS: Windows
- Embedded Workbench: ARM
- Embedded Workbench version: 9.30.1 and 8.32.1 are both installed.
- VSC Extension version: Latest at time of writing (v1.30.4)

**Additional context:**
Add any other context about the problem here, e.g., projects, files etc. Logs for the debugger can be obtained from the debug console and trace can activated by adding *"trace": true* to the launch configuration.

jlonnberg, 2024-01-31:
@cosberg-rzl Thanks for the report.
We've added the full support for attach to running target in EWARM 9.40, which is primarily related to the leave-target-running command to avoid an unwanted stop when a debug session ends.

- Which driver, e.g., I-Jet or JLink, are you using for the project? Can you provide use the launch.json?
- Moreover, are you running with the leave-target-running option set during the first session to avoid terminating the target?
- Does the same thing work in the Embedded Workbench?

cosberg-rzl, 2024-01-31:
> Which driver, e.g., I-Jet or JLink, are you using for the project? Can you provide use the launch.json?
It's an I-Jet. Here's my launch.json, with some sensitive paths edited out:
https://hastebin.skyra.pw/abemadiqes.swift

> Moreover, are you running with the leave-target-running option set during the first session to avoid terminating the target?
Yep, as far as I know. It does seem to correctly attach at least.

> Does the same thing work in the Embedded Workbench?
Yeah, we don't have any problems in the IAR IDE with any of the debugging options.

jlonnberg, 2024-02-01:
So, I cannot see anything strange here. 
Can you please share the debug session output for a session where the problem occurs?
Moreover, can you start the session with the option "--drv_communication_log=./cspycomm.log" added and send us the log-file?

closed 2026-09-29 by jlonnberg