iarsystems/iar-vsc-debug #37: Resetting the target not always successful when using preLaunchTask

hyperspacex2, 2024-10-04:
**Describe the bug:**
Selecting "Restart" crashes debugging session after showing toast "Failed to reset target" if using a preLaunchTask.

**To Reproduce:**
* Use a preLaunchTask in launch config
* Start debug session
* Pause
* Restart

**Expected behavior:**
Target resets after preLaunchTask was performed. Sometimes this works as expected but often it fails to reset.

**Actual behavior:**
A toast "Failed to reset target" is shown. After that the debug session is frozen.

**Environment:**

- OS: Windows
- Embedded Workbench: Arm
- Embedded Workbench version: 9.4.1
- VSC Extension version: v1.30.4
- C-SPY version: 9.2.2.10770

**Additional context:**
To my knowledge vs code has no option to have a task which is only executed pre launch and not upon restart.
The prelaunch task in my configuration executes the iar build task. That itself is not ideal, however since I would like to restart usually build has no changes so a rebuild is quick.

"--jlink_reset_strategy" <= has no effect on the behaviour described

Used driverOptions in launch configuration:

"driverOptions": [
	"--crun=disabled",
	"--endian=little",
	"--cpu=Cortex-M7",
	"--fpu=VFPv5_D16",
	"-p",
	"$TOOLKIT_DIR$\\config\\debugger\\NXP\\MIMXRT1062xxx6B.ddf",
	"--semihosting",
	"--device=MIMXRT1062xxx6B",
	"--drv_communication_log=C:\\GitProjects\\myproject\\iar\\cspycomm.log",
	"--drv_communication=TCPIP:xxxxxxxx",
	"--drv_interface_speed=20000",
	"--drv_default_breakpoint=1",
	"--drv_restore_breakpoints=_call_main",
	"--jlink_reset_strategy=0,2",
	"--drv_interface=SWD",
	"--drv_catch_exceptions=0x000",
	"--drv_swo_clock_setup=117333000,0,40000000",
	"--jlink_device_name=MIMXRT1062xxx6B",
],

jlonnberg, 2024-10-07:
Hi @hyperspacex2,
Does the reset work if the you exclude the pre-build task?

hyperspacex2, 2024-10-07:
> Hi @hyperspacex2, Does the reset work if the you exclude the pre-build task?

Hi @jlonnberg. Yes I can confirm that without any pre launch task the reset does work.

jlonnberg, 2024-10-07:
@hyperspacex2 
I've done some digging and this is not a problem with our extension but rather how vscode does things. A prelaunchtask is executed before starting or restarting, which is not ideal for some embedded situations. This is not something that we control and the problem has been raised before:
https://github.com/microsoft/vscode/issues/136384
https://github.com/ARM-software/vscode-embedded-debug/issues/2

The reasoning behind the behavior is related to how, for instance, node developers want the debugger to behave. I'll have some internal discussions about perhaps exposing the restart command in some other way that helps your case, but the restart-support from microsoft will not work unless the pre-launch task is removed

jlonnberg, 2024-10-07:
@hyperspacex2 
We'll include a WA for this in the upcoming release, so that you'll be able to run a command for restart to avoid the prelaunch-task :)

hyperspacex2, 2024-10-08:
> @hyperspacex2 I've done some digging and this is not a problem with our extension but rather how vscode does things. A prelaunchtask is executed before starting or restarting, which is not ideal for some embedded situations. This is not something that we control and the problem has been raised before: [microsoft/vscode#136384](https://github.com/microsoft/vscode/issues/136384) [ARM-software/vscode-embedded-debug#2](https://github.com/ARM-software/vscode-embedded-debug/issues/2)
> 
> The reasoning behind the behavior is related to how, for instance, node developers want the debugger to behave. I'll have some internal discussions about perhaps exposing the restart command in some other way that helps your case, but the restart-support from microsoft will not work unless the pre-launch task is removed

Yes I've seen that. I suppose vs code needs to support different use cases.

hyperspacex2, 2024-10-08:
> @hyperspacex2 We'll include a WA for this in the upcoming release, so that you'll be able to run a command for restart to avoid the prelaunch-task :)

Sounds great. Thank you!

jlonnberg, 2024-10-17:
New reset command included in the upcoming 1.40 release. From the command set, run "IAR Debug: Reset the session without running prelaunch task" to reset the target directly

closed 2024-10-17 by jlonnberg