iarsystems/iar-vsc-debug #38: Pressing the restart button without pausing breaks the debug session

FT9R, 2024-12-24:
**Describe the bug:**
When i hit restart button (Ctrl+Shift+F5) without pausing the debug session the debugger becomes no longer available.

**To Reproduce:**
* Start debug session (F5)
* Press continue (F5)
* Press restart (Ctrl+Shift+F5)

**Expected behavior:**
The application entry point is reached and debug session is paused.

**Actual behavior:**
Debugger does not respond to any actions.
Debug console content: `Reset while executing is not supported by this target`.
VSCode messages: `Failed to go to specified location`.

**Environment:**
- OS: Windows_NT x64 10.0.19045
- Embedded Workbench: Arm
- Embedded Workbench version: 9.2
- VSC Extension version: 1.40.1

HampusAdolfsson, 2025-01-07:
Hi, thanks for reporting this!

As pointed out by the log message, not all debugger drivers support restarting the program while it is executing. This is the same when debugging in Embedded Workbench and it is not something we can address from this VS Code extension.

However, failing to restart should not cause the debugger to become unresponsive. I've tracked this down to a bug in VS Code which I've reported here:
https://github.com/microsoft/vscode/issues/237414
Once it is fixed, the debugger should stay responsive after a failed restart.

FT9R, 2025-01-21:
@HampusAdolfsson Hello. Thank you for the response. Could you then make the restart button inactive when debug session is not paused for these types of drivers? Or is it strictly related to VS Code?

HampusAdolfsson, 2025-05-02:
@FT9R I've added a workaround for this issue, so the session should stay responsive after a failed restart. By telling VS Code that the target was successfully restarted even if it wasn't, we can avoid triggering the bug. 

The workaround will be included in the next release, within the next month or two.